microsoft/psi #114: Broken Link -  Microsoft Speech Platform SDK v11.0.

sidshenoy93, 2021-03-18:
I noticed three broken links on the "Building the Codebase" page. 

1. The text "Microsoft Speech Platform SDK v11.0." points to a broken link: [](http://go.microsoft.com/fwlink/?LinkID=223570)
2. The text "Microsoft Speech Platform Runtime v11.0" points to a broken link: [](http://go.microsoft.com/fwlink/?LinkID=223568)
3. The text "Language Pack" points to a broken link: [](http://go.microsoft.com/fwlink/?LinkID=223569)

sandrist, 2021-03-19:
Hi Sid, thanks for pointing that out! It looks like those links might have gotten broken sometime recently. We're looking into it now to see if we can dig up fresh links for those resources, and will update the instructions accordingly.

In the meantime, bear in mind that these downloads are optional, and not required for building the main codebase.

[This \psi component](https://github.com/microsoft/psi/tree/master/Sources/Speech/Microsoft.Psi.Speech.Windows) also has speech recognition and synthesis functionality, but uses the [System.Speech](https://docs.microsoft.com/en-us/dotnet/api/system.speech.recognition) library rather than the Microsoft Speech Platform SDK (they are similar but subtly different).

chitsaw, 2021-06-15:
The links have been permanently removed and these downloads are no longer officially available. We will probably deprecate the Microsoft.Psi.MicrosoftSpeech.Windows component in an upcoming release.