microsoft/psi #117: Plotting FFT Response of Input Audio

BryanT423, 2021-03-27:
Hello all, I am new to C# and am trying to display a live FFT response of my input microphone audio. I would like to display the response on an instant 2D Panel. I understand that I should enable 3rd Party Visualizers for access to LiveCharts [https://lvcharts.net/](url),  but I am still unable to plot the FFT after I write it into my store. I think the trouble I am having is related to the DLL files when I try to add them into my PsiStudio.xml file. I'm currently adding `Microsoft.Psi.LiveCharts.Visualization.Windows.dll` `Microsoft.Psi.Visualization.Windows.dll` `LiveCharts.dll` and `Microsoft.Psi.Audio.dll`

I believe some of my confusion is coming from understanding which DLL file will help me plot the cartesian chart needed for my FFT plot.

How can I tell which visualizer/.dll file within the `Microsoft.Psi.LiveCharts.Visualization.Windows` library is necessary for my use?

Any and all help would be greatly appreciated.

Here is the code for my program.cs file:

                // Create the AudioCapture component and write the output to the store
                var audio = new AudioCapture(p, new AudioCaptureConfiguration() 
                { 
                    Format = WaveFormat.Create16kHz1Channel16BitPcm()
                });
                audio.Write("Audio", store);

                AcousticFeaturesExtractorConfiguration Default = new AcousticFeaturesExtractorConfiguration();
                var acousticFeaturesExtractor = new AcousticFeaturesExtractor(p, Default);
                audio.PipeTo(acousticFeaturesExtractor);

                //Compute FFT
                acousticFeaturesExtractor.FFT
                    .Sample(TimeSpan.FromSeconds(0.2))
                    .Select(fft =>
                    {
                        var labeledValues = new List<(string, double)>();

                        for (int i = 0; i < fft.Length; i++)
                        {
                            labeledValues.Add((i.ToString(), fft[i]));
                        }
                        return labeledValues;
                    })
                    .Write("FFT", store);

BryanT423, 2021-03-31:
I do not see the dialog box. I realized that I was in the PsiStudio.xml file instead of the PsiStudioSettings.xml. Although I've looked in my PsiStudio folder and I can't seem to find the PsiStudioSettings.xml, specified in the 3rd party visualizers. Only PsiStudioSettings.cs.
Am I missing something I should have installed or am I looking in the wrong place?

[image]

[image]

Here is the path I am referencing in my PsiStudio.xml file, although I will need to move it to PsiStudioSettings.xml
`<AdditionalAssemblies>C:\Users\bryan\source\repos\psi12\Sources\Visualization\Microsoft.Psi.LiveCharts.Visualization.Windows\bin\Debug\net472\Microsoft.Psi.LiveCharts.Visualization.Windows.dll</AdditionalAssemblies>`

BryanT423, 2021-03-31:
Hi Sean, it works now! I had to look for the PsiStudioSettings.xml outside of my current project. I just searched for the PsiStudio folder in my entire C drive and found it. Thanks a lot!
[image]

closed 2021-03-31 by BryanT423

sandrist, 2021-03-31:
Great, glad it works!