microsoft/psi #118: Exception in Microsoft.Psi.Onnx when running in net472

xiangzhi, 2021-04-01:
I was experimenting with the ONNX component and found it was not working in `net472` applications. The exact same application works when I change it to `netcoreapp3.1`. 

```
System.TypeInitializationException
  HResult=0x80131534
  Message=The type initializer for 'Microsoft.ML.OnnxRuntime.NativeMethods' threw an exception.
  Source=Microsoft.ML.OnnxRuntime
  StackTrace:
   at Microsoft.ML.OnnxRuntime.SessionOptions..ctor()
   at Microsoft.ML.OnnxRuntime.InferenceSession..ctor(String modelPath)
   at Microsoft.ML.Transforms.Onnx.OnnxModel..ctor(String modelFile, Nullable`1 gpuDeviceId, Boolean fallbackToCpu, Boolean ownModelFile, IDictionary`2 shapeDictionary)
   at Microsoft.ML.Transforms.Onnx.OnnxTransformer..ctor(IHostEnvironment env, Options options, Byte[] modelBytes)
   at Microsoft.ML.Transforms.Onnx.OnnxTransformer..ctor(IHostEnvironment env, String[] outputColumnNames, String[] inputColumnNames, String modelFile, Nullable`1 gpuDeviceId, Boolean fallbackToCpu, IDictionary`2 shapeDictionary)
   at Microsoft.ML.OnnxCatalog.ApplyOnnxModel(TransformsCatalog catalog, String[] outputColumnNames, String[] inputColumnNames, String modelFile, Nullable`1 gpuDeviceId, Boolean fallbackToCpu)
   at Microsoft.Psi.Onnx.OnnxModel..ctor(OnnxModelConfiguration configuration) in C:\Users\Zhi\source\repos\CMU-TBD\psi\Sources\Integrations\Onnx\Common\OnnxModel.cs:line 44

Inner Exception 1:
DllNotFoundException: Unable to load DLL 'onnxruntime': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
```

This is my `csproj` file for reference:
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>net472</TargetFramework>
    <OutputType>exe</OutputType>
  </PropertyGroup>

  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|AnyCPU'">
    <PlatformTarget>AnyCPU</PlatformTarget>
  </PropertyGroup>

  
  <ItemGroup>
    <ProjectReference Include="..\..\Sources\Integrations\Onnx\Microsoft.Psi.Onnx.ModelRunners.Cpu\Microsoft.Psi.Onnx.ModelRunners.Cpu.csproj" />
  </ItemGroup>
</Project>

```

chitsaw, 2021-04-05:
Have you tried changing the `<PlatformTarget>` to `x64`?

xiangzhi, 2021-04-06:
Cool. It worked! I really should remember to try that first if I ever run into compiling/building issues.

closed 2021-04-06 by xiangzhi