microsoft/psi #127: [Bug] 3rd party Stream Readers visualization not work in latest version of PsiStudio

xiangzhi, 2021-04-24:
In the latest version of PsiStudio, when I try to import/open a WAVE file. The file opens correctly and shows the correct meta data on the side bar. However, when I try visualizing the stream, the following error is thrown:
```
System.InvalidOperationException
  HResult=0x80131509
  Message=No matching files found: C:\Users\Zhi\Downloads \[gettysburg.wav.*\]gettysburg.wav*
  Source=Microsoft.Psi
  StackTrace:
   at Microsoft.Psi.PsiStore.GetPathToLatestVersion(String storeName, String rootPath) in C:\Users\Zhi\source\repos\CMU-TBD\psi\Sources\Runtime\Microsoft.Psi\Data\PsiStore.cs:line 496
```
I originally tried it with our lab's stream readers which also throw a similar error. It is possible that this error might be unique to my setup and was cause by the update process.

chitsaw, 2021-04-24:
Looking into this now... When you say latest version of PsiStudio, do you mean you just synced with the latest sources that were pushed yesterday? If so, what happens if you sync to the version just before? Does that resolve the error and get you unblocked for now?

xiangzhi, 2021-04-25:
Thanks for looking into this Nick! Sorry about that, I should have mention the exact version. 

Yes, this was the version (0.15) that was released yesterday. I merged in the latest changes and rebuilded PsiStudio from scratch. The problem only shows up when you use third party readers.

I was worried it might be caused by low level changes I made to Psi in our own branches but I just checkouted the origin main branch, updated it to (0.15), and the bug is still there. The issue was easily reproducible:
1. Start PsiStudio
2. Open a WAVE file using the built-in provided WAVE file reader.
3. Try visualize the `Audio` stream.
4. PsiStudio should crash at this point

I'm testing all these on a separate test branch right now to see the cool stuff y'all added (Yeah for clear layout button 🎉). I'm working through other parts of the update right now but can easily revert back if needed 🥳

PR #128 (2021-04-25): Don't simplify StreamReaderTypeName
Fixes #127
closed 2021-04-25 by chitsaw