microsoft/psi #129: Echo cancellation

KanaHayama, 2021-04-26:
I want to record audio with AudioCapture, but it also records the sound that the computer is playing.
Will there be a component or an AudioCapture option for echo cancellation?
Thanks.

sandrist, 2021-04-26:
If you are using a Windows `AudioCapture` component, you can set `OptimizeForSpeech` to true in the `AudioCaptureConfiguration` that you pass to the constructor of the component, which should turn on echo cancellation. But this might not work for all devices, unfortunately.

```csharp
/// <summary>
/// Gets or sets a value indicating whether the captured audio should be pre-processed for
/// speech recognition applications. Available in Windows 10 Creators Update or later.
/// </summary>
/// <remarks>
/// In later versions of Windows 10, the audio capture pipeline optionally enables additional
/// signal processing to apply enhancements such as echo and noise suppression to optimize
/// the audio for speech recognition applications. By default, this option is set to false.
/// This feature may not be available for all capture devices.
/// </remarks>
public bool OptimizeForSpeech { get; set; } = false;
```

KanaHayama, 2021-04-26:
Thank you. That option does not work very well for my case. I will find a more advanced solution.

closed 2021-04-26 by KanaHayama