microsoft/psi #142: Can't execute the SimpleVoiceActivityDetector on Ubuntu

hlahkar, 2021-05-05:
I am on Ubuntu 20.04; I get the below exception on executing the SimpleVoiceActivityDetector

Unhandled exception. System.ArgumentException: Hardware set params failed (ALSA error code: -12).
   at Microsoft.Psi.Audio.LinuxAudioInterop.CheckResult(Int64 result, String message)
   at Microsoft.Psi.Audio.LinuxAudioInterop.Open(String name, Mode mode, Int32 rate, Int32 channels, Format format, Access access)
   at Microsoft.Psi.Audio.AudioCapture.Start(Action`1 notifyCompletionTime)
   at Microsoft.Psi.Executive.PipelineElement.<Activate>b__46_0()
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context)
   at Microsoft.Psi.Scheduling.Scheduler.Run(Object workItem)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()
Aborted

AshleyF, 2021-05-06:
My first guess is that the default ALSA parameters are incorrect for your hardware. Try listing your devices with `arecord -l` and note the card/device numbers. The default `deviceName` used by `AudioCapture` (line 25 of Program.cs) is `"plughw:0,0"`, meaning card 0, device 0. Other parameters may also be incompatible with your hardware. The samples uses 16kHz, single-channel PCM, but perhaps try other `WaveFormats`.

Settings can also be tested with the ALSA record tool, e.g. `arecord -D pludhw:0,0 -r 16000 -c 1 -d 3`, meaning card 0, device 0, 16kHz, 1-channel, 3 second test recording.

Good luck, and let us know if this help!

hlahkar, 2021-05-06:
I ran the commands as mentioned; got the following result

(base) himangshu@himangshu:~/work/microsoft/psi-samples/Samples/SimpleVoiceActivityDetector/bin/Debug/netcoreapp3.1$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC294 Analog [ALC294 Analog]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

(base) himangshu@himangshu:~/work/microsoft/psi-samples/Samples/SimpleVoiceActivityDetector/bin/Debug/netcoreapp3.1$ arecord -D pludhw:0,0 -r 16000 -c 1 -d 3
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM pludhw:0,0
arecord: main:830: audio open error: No such file or directory

AshleyF, 2021-05-06:
Sorry, that was a typo. It’s `plughw` rather than `pludhw`.

hlahkar, 2021-05-06:
arecord -D plughw:0,0 -r 16000 -c 1 -d 3 - this command works fine. But I am still getting the exception in the App

AshleyF, 2021-05-06:
I setup an Ubuntu 20.04 machine and can repro what you see. Could you try changing the sampling rate? I repro at 16kHz, but not at 44.1kHz, for example. I also repro that both work with `arecord` but only the latter works with the \psi `AudioSource` component. I'm still trying to get to the bottom of why that would be...

```
var microphone = new AudioCapture(p, WaveFormat.Create16BitPcm(44100, 1), "plughw:0,0");
```

hlahkar, 2021-05-07:
> I setup an Ubuntu 20.04 machine and can repro what you see. Could you try changing the sampling rate? I repro at 16kHz, but not at 44.1kHz, for example. I also repro that both work with `arecord` [...]

This change works. Thanks for the support.

AshleyF, 2021-05-07:
Glad that worked. It seems that `arecord` silently resamples as needed. By the way, if you want to similarly down sample to 16kHz in \psi, this can be done with `microphone.Resample(WaveFormat.Create16kHz1Channel16BitPcm())`.

Have fun, and don't hesitate to come back with any other stumbling blocks you hit.

closed 2021-06-15 by AshleyF