microsoft/psi #143: PlatformNotSupportedException when opening Windows-saved store on Ubuntu Linux

svenvdvoort, 2021-05-05:
Hello everyone,

We're having trouble opening a PsiStore file stored using a Windows \psi application with a \psi application running under Ubuntu Linux.

The code that was used to create the store on Windows 10 is as follows:
```
var pipeline = Pipeline.Create();
var video_store = PsiStore.Create(pipeline, "VideoRawData", PSI_STORES_PATH);

var webcam = new MediaCapture(pipeline, 960, 540, 30);
webcam.Write("Webcam", video_store);

pipeline.Run();
```

The code that is used to display the video stream again on Ubuntu 20.04:
```
using (var p = Pipeline.Create())
{
    var store = PsiStore.Open(p, "VideoRawData", PSI_STORES_PATH);

    var webcam = store.OpenStream<double>("Webcam");
}
```

Running the latter program under Linux, the second statement throws an exception:
```
Unhandled exception. System.PlatformNotSupportedException: Named maps are not supported.
   at System.IO.MemoryMappedFiles.MemoryMappedFile.OpenCore(String mapName, HandleInheritability inheritability, MemoryMappedFileRights rights, Boolean createOrOpen)
   at System.IO.MemoryMappedFiles.MemoryMappedFile.OpenExisting(String mapName, MemoryMappedFileRights desiredAccessRights, HandleInheritability inheritability)
   at Microsoft.Psi.Persistence.InfiniteFileReader.LoadNextExtent()
   at Microsoft.Psi.Persistence.InfiniteFileReader.MoveNext()
   at Microsoft.Psi.Persistence.MetadataCache.Update()
   at Microsoft.Psi.Persistence.MetadataCache..ctor(String name, String path, Action`2 entriesAdded)
   at Microsoft.Psi.Persistence.PsiStoreReader..ctor(String name, String path, Action`2 metadataUpdateHandler, Boolean autoOpenAllStreams)
   at Microsoft.Psi.Data.PsiStoreStreamReader..ctor(String name, String path)
   at Microsoft.Psi.Data.PsiImporter..ctor(Pipeline pipeline, String name, String path, Boolean usePerStreamReaders)
   at Microsoft.Psi.PsiStore.Open(Pipeline pipeline, String name, String rootPath, Boolean usePerStreamReaders)
   at PSIScripts.DisplayRecording.Program.Main(String[] args) in /<redacted>/DisplayRecording/Program.cs:line 26
```

This does not happen if I record webcam data using a (Linux-specific) script and then read it on the same Linux machine. This leads me to think that \psi on Linux has trouble reading a store generated using \psi on Windows. According to the documentation this should not be a problem however, since the \psi platform and SDK should be cross-platform compatible.

We would really appreciate any help on this issue! :)

xiangzhi, 2021-05-05:
Hi,

I notice that when you opening it on the Linux side, you are opening it as a stream of `double`. I believe  `MediaCapture` saves it as a stream of `Shared<Image>`. Can you try the following code instead and see if it works?
```
using (var p = Pipeline.Create())
{
    var store = PsiStore.Open(p, "VideoRawData", PSI_STORES_PATH);

    var webcam = store.OpenStream<Shared<Image>>("Webcam");
}
```

svenvdvoort, 2021-05-05:
You are correct! This was done in an effort to debug the problem. It accidentally ended up in the GitHub issue.
The exact code @xiangzhi provides gives the same error, indicating that is does not have to do with the type of data stream being opened.

xiangzhi, 2021-05-05:
It's possible you ran into the same problem I had in #74 just in reverse (mine was from Ubuntu to Windows). I remember @chitsaw saying parts of it were fixed but I'm not sure if it also extend to `Shared<Image>`. 

What frameworks you are using in each program and what are their versions?

chitsaw, 2021-05-06:
Hi @svenvdvoort - I am unable to repro your issue, but from the stack trace, one possibility is that the `InfiniteFileReader` was created with a null `path` argument. I don't see how that exception would have been hit otherwise. Named memory-mapped files are not supported on Linux, and this is the exception you're hitting, but that code path is only taken if `path == null`. Could you confirm the value of your `PSI_STORES_PATH` variable just before calling `PsiStore.Open`?

svenvdvoort, 2021-05-08:
Hello, thanks for your help! I think this solved the problem.
The `PsiStore.Open(...)` method could use a null-check in this case :)

I did get another issue however: the video output on Linux looks very blue-ish, which it does not on Windows. I will investigate this further, but it's probably material for a different issue.

closed 2021-06-15 by chitsaw