microsoft/psi #153: Question : Conversion from .psi to .csv

Nazehyounsi, 2021-05-27:
Hi, 

I would like to know if it is possible to convert data stored as Binary files in a Psi store, to Csv files using Psi studio.

I have several data extracted from Openface, exported in a Psi store with .psi extension. I imported these data on Psi studio but I can't find a way to read them or directly convert them to Csv files.

Do you know please how I could  achieve this task ? 

Thank you for your help and attention.

AshleyF, 2021-05-27:
We do have an [interop infrastructure](https://github.com/microsoft/psi/tree/master/Sources/Runtime/Microsoft.Psi.Interop) allowing you to read/write \psi streams in formats other than \psi stores. This [includes CSV](https://github.com/microsoft/psi/blob/master/Sources/Runtime/Microsoft.Psi.Interop/Format/Readme.md#concrete-csv), both over a [network transport or to/from flat files](https://github.com/microsoft/psi/blob/master/Sources/Runtime/Microsoft.Psi.Interop/Transport/Readme.md). Additionally, the PsiStoreTool provides a command-line interface for [saving/sending streams from a \psi store](https://github.com/microsoft/psi/blob/master/Sources/Tools/PsiStoreTool/Readme.md#saving) (e.g. `PsiStoreTool save -p E:\Data -d MyStore -s MyStream -f MyData.csv -m csv`).

Hopefully that helps! Let us know if you run into issues.

xiangzhi, 2021-05-27:
Ashley gave a great answer, and here's a more hacky alternative. 

I'm not familiar with how OpenFace works and how their store their data. The tricky part is that there is no one right way to convert each data stream to a CSV (e.g. some people might want to ignore the timing information, others don't want to). In our own work, we end up just writing a simple program open the store, read the data ,and save it into a text file with CSV format. There is an added benefit that the CSV will always be in the format you want and isn't limited to just one data source.  Here's a lightly edited example form our code.
```csharp
using (var p = Pipeline.Create(enableDiagnostics: true))
{
    var inputStore = PsiStore.Open(p, "store_name", "store_path");
    
    csSource = inputStore.OpenStream<CoordinateSystem>("cs");
    var csv_lines = new List<string>();
    csSource.Do(m =>
    {
        csv_lines.Add($"first_column,{String.Join(",", m.Storage.ToRowMajorArray())}");
    });

    p.Run(ReplayDescriptor.ReplayAll);

    System.IO.File.WriteAllLines("example.csv", csv_lines);
}
```

Hope this helps!

AshleyF, 2021-06-15:
Hopefully one of these solutions worked for you. Please feel free to reopen this or open new issues if we can be of more help.

closed 2021-06-15 by AshleyF