microsoft/psi #154: How to order the data frames in multiple streams with different data types by their OriginatingTime

KanaHayama, 2021-05-27:
I have two streams, with different data types. One stream is delayed, so they are mis-aligned.
Now I want to process data in both streams in chronological order, how can I achieve that?

I read the document and found, the `Join()` operator isn't suitable, because it may return data frames in the secondary stream more than one time. `Join()` with `Reproducible.ExactOrDefault<T>()` isn't suitable, because data frames in the secondary stream without an exactly time match will not be returned.

KanaHayama, 2021-05-27:
Now I am trying to using this approach

```C#
var time1 = InConnector.Select((_, e) => e.OriginatingTime);
var time2 = AudioInConnector.Select((_, e) => e.OriginatingTime);
var time = time1.Zip(time2);
var join1 = time.Join(InConnector, Reproducible.ExactOrDefault<T1>());
var join2 = join1.Join(AudioInConnector, Reproducible.ExactOrDefault<T2>());
var join = join2.Select(data => (data.Item2, data.Item3));
```

Is there a handy approach for this?

sandrist, 2021-05-27:
You should be able to Zip the two original streams directly, without first selecting out the originating times and re-joining everything back together.

Zip outputs all messages across the input streams in merged originating time order.

(The reason Zip outputs an array is in case it receives multiple inputs with the same originating time, in which case they are emitted together as one message in an array. Otherwise, the output array will usually just have one element in it.)

KanaHayama, 2021-05-27:
@sandrist, Thank you for your reply.
I cannot find a Zip overload accepting two streams with different data types. I am using the latest 0.15.49.1-beta.

KanaHayama, 2021-05-27:
Oh, I found it and it is not an extension method

sandrist, 2021-05-27:
I see, yeah we don't have a component for merge/zip of streams with heterogenous data types. The solution you came up with originally looks good though, and I would advise simply wrapping it into your own extension method to reuse easily.

KanaHayama, 2021-05-27:
Thank you.
If /psi provides a shortcut for this kind of operation in the future, it can be very helpful.

closed 2021-05-27 by KanaHayama