microsoft/psi #156: RealSense example

profwillie, 2021-06-03:
Is there a working example of the RealSense psi component?  I know that @kahlilfitz has an example, but that example creates a new component, and I'm hoping to use the component here: https://github.com/microsoft/psi/tree/master/Sources/RealSense

One problem I am having to start:  Is the code available in a nuget?  Or do I need to use the source for this?

sandrist, 2021-06-04:
Hi Willie, unfortunately we are not able to generate a nuget for that one, because we can't redistribute the [RealSense SDK](https://www.intelrealsense.com/developers/) that it depends on. To use that component, you'll need to build the project from source. Thankfully, in addition to the native `Microsoft.Psi.RealSense_Interop.Windows.x64` project, it only depends on `Microsoft.Psi` and `Microsoft.Psi.Image`. In addition, you'll need to separately download the [RealSense SDK](https://www.intelrealsense.com/developers/) somewhere, and create an environment variable on your machine called "RealSenseSDKDir" that points to it.

trigaten, 2021-06-16:
@profwillie 
Here is a working code example (assuming you have your dependencies and PATHs in order):

```var p = Pipeline.Create();
            RealSenseSensor r = new(p);
            const int widthOutput = 160;
            const int heightOutput = 90;

            StringBuilder sb = new StringBuilder();
            r.ColorImage.Resize(widthOutput, heightOutput).Do(color =>
            {
                var bitmap = color.Resource.ToBitmap();

                // render color frame as "ASCII art"
                sb.Clear();
                for (int y = 0; y < heightOutput; y += 2)
                {
                    for (int x = 0; x < widthOutput; x++)
                    {
                        var p = bitmap.GetPixel(x, y);
                        sb.Append(" .:-=+*#%@"[(int)((p.R + p.G + p.B) / 76.5)]);
                    }

                    sb.Append(Environment.NewLine);
                }

                Console.SetCursorPosition(0, 0);
                Console.WriteLine(sb.ToString());
            });

            Console.BackgroundColor = ConsoleColor.Black;
            Console.ForegroundColor = ConsoleColor.White;
            Console.Clear();
            p.RunAsync();
            Console.ReadLine();
```