microsoft/psi #163: Is there a method to introduce latency to a stream?

KanaHayama, 2021-06-09:
I tried `Microsoft.Psi.Operators.Delay` and found it will offset the timestamp which does not meet my requirements.

I have 2 streams, A and B.
+ Stream A is a realtime video stream.
+ Stream B has command signals, this stream has latency within a range, and the time interval of adjacent signals can be large.

I want to fuse them with `Available.Last()`, so introducing a small latency to stream A will help accommodate latency in stream B.
Is there a method doing something like this?
```C#
public static partial class Operators {
    public static IProducer<T> IntroduceLatency<T>(this IProducer<T> source, TimeSpan latency, DeliveryPolicy<T> deliveryPolicy = null) {
        return source.Process<T, T>((d, e, s) => {
            _ = Task.Factory.StartNew(async () => {
                await Task.Delay(latency);
                s.Post(d, e.OriginatingTime);//this approach, post order may not correct
            });
        });
    }
}
```

sandrist, 2021-06-09:
Do the messages in Stream B have valid originating times, so the latency is known to Psi? You might want to look into the collection of Reproducible interpolators available in `Reproducible.cs`, where matching is performed based on originating time rather than wall-clock time, which helps make your fusion robust to differences in latencies across the two streams. The `Available` interpolators make matches greedily based on wall clock time of when messages are delivered, but you can get into trouble when latencies differ.

If you truly want to inject latency into a stream, I don't believe we have any operators for that, but it would be fairly easy to accomplish with something like a Thread sleep inside a `Do` operator.

KanaHayama, 2021-06-16:
Thank you. The stream B has a valid originating time. I am not using Reproducible because I am afraid that it may block and buffer all video frames if no signal come from stream B.
My concern is that OS thread scheduling may not strictly follow the order messages are posted, so I want to find another order guaranteed approach. I will try to implement one myself if it is not planned in /psi future release.

closed 2021-06-16 by KanaHayama