microsoft/psi #164: psiStudio freezes when saving annotations

iboucher125, 2021-06-09:
Hello,

I am trying to create time interval annotations. I have the finite annotation schema definition saved and then I opened a store inside psiStudio that contains both audio and video. I then created a new annotation stream, selected "create in new partition" and created a new a store for the annotations. I was able to add annotations but psiStudio freezes when I try to save the annotations stream. The saving store progress bar is displayed but there is no progress.  Also there are three new folders (BeforeEdit, Edit, Edit) in the directory containing the store with the annotations.

skdent, 2021-06-09:
Hi iboucher125

I'd like to investigate this issue further.  If you're running PsiStudio under the debugger in Visual Studio, then when PsiStudio freezes could you hit the Break button in Visual Studio and then copy the call stack, as that will give me more data to work with.

skdent, 2021-06-10:
It appears that we're not correctly releasing the store files before we try to copy them to the backup directory.  I'm working on a fix for you now.

iboucher125, 2021-06-10:
We ended up running PsiStudio under the debugger and then recompiled and ran it in release mode without making any changes and it worked.

closed 2021-06-10 by iboucher125