microsoft/psi #165: [Bug] Unable to compile from source for all unix systems.

xiangzhi, 2021-06-10:
I was trying to compile the source code from scratch when I ran into the following build error:
```
 
Welcome to .NET Core 3.1!
---------------------
SDK Version: 3.1.410

Telemetry
---------
The .NET Core tools collect usage data in order to help us improve your experience. It is collected by Microsoft and shared with the community. You can opt-out of telemetry by setting the DOTNET_CLI_TELEMETRY_OPTOUT environment variable to '1' or 'true' using your favorite shell.

Read more about .NET Core CLI Tools telemetry: https://aka.ms/dotnet-cli-telemetry

----------------
Explore documentation: https://aka.ms/dotnet-docs
Report issues and find source on GitHub: https://github.com/dotnet/core
Find out what's new: https://aka.ms/dotnet-whats-new
Learn about the installed HTTPS developer cert: https://aka.ms/aspnet-core-https
Use 'dotnet --help' to see available commands or visit: https://aka.ms/dotnet-cli-docs
Write your first app: https://aka.ms/first-net-core-app
--------------------------------------------------------------------------------------
Microsoft (R) Build Engine version 16.7.2+b60ddb6f4 for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj (in 1.28 sec).
  Assembling 'IL/MemoryAccess.il' , no listing file, to dll --> 'bin/Debug/netstandard2.0/Microsoft.Psi.IL.dll'
  
  Operation completed successfully
Data/Importer.cs(111,68): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(273,72): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(369,42): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(386,42): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]

Build FAILED.

Data/Importer.cs(111,68): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(273,72): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(369,42): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
Data/Importer.cs(386,42): error CS8652: The feature 'not pattern' is currently in Preview and *unsupported*. To use Preview features, use the 'preview' language version. [/home/xiangzht/Dev/csharp/psi/Sources/Runtime/Microsoft.Psi/Microsoft.Psi.csproj]
    0 Warning(s)
    4 Error(s)

Time Elapsed 00:00:02.45
``` 
I actually started on a Mac system but the print out above came from a 20.04 Ubuntu System. The error seemed to be coming from the usage `is not` pattern in `Microsoft.Psi.Data.Importer.cs`.
```csharp
if (this.streamReader.GetStreamMetadata(streamName) is not PsiStreamMetadata psiStreamMetadata)
```
As far as I can tell the `is not` pattern was only included in C# 9.0 which is not part of .NET standard 2.0. I'm not sure why it didn't throw an error on Windows machines.

xiangzhi, 2021-06-10:
Upon more testing, the compiling would work if I have the `.NET Core 5.0` sdk installed. I'm still puzzled by the fact that we can set a package as `netstandard2.0` but use C# 9 features.

chitsaw, 2021-06-15:
Ah, that's because C# 9 is a language (compiler) feature which you get when you install the new .NET5 SDK, but that is independent of the framework you're targeting (e.g. `netstandard2.0`).

xiangzhi, 2021-06-15:
Make sense! I guess we should edit the wiki/readme to change it to .NET5 SDK :)

chitsaw, 2021-06-15:
> Make sense! I guess we should edit the wiki/readme to change it to .NET5 SDK :)

Done!

closed 2021-06-15 by chitsaw