microsoft/psi #170: Cant load Microsoft.Psi.RealSense_Interop.Windows.x64.dll

trigaten, 2021-06-15:
I am trying to use the RealSense component, but I get the following error: 
```
System.IO.FileNotFoundException: 'Could not load file or assembly 'Microsoft.Psi.RealSense_Interop.Windows.x64.dll' or one of its dependencies. The specified module could not be found.'
```

I have built from source following the Wiki instructions for supporting RealSense.

Here is my .csproj:
```
<Project Sdk="Microsoft.NET.Sdk">

	<PropertyGroup>
		<OutputType>Exe</OutputType>
		<TargetFramework>net472</TargetFramework>
		<Platforms>x64</Platforms>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">
		<PlatformTarget>x64</PlatformTarget>
	</PropertyGroup>

	<ItemGroup>
		<ProjectReference Include="..\Sources\Imaging\Microsoft.Psi.Imaging\Microsoft.Psi.Imaging.csproj" />
		<ProjectReference Include="..\Sources\RealSense\Microsoft.Psi.RealSense.Windows.x64\Microsoft.Psi.RealSense.Windows.x64.csproj" />
		<ProjectReference Include="..\Sources\RealSense\Microsoft.Psi.RealSense_Interop.Windows.x64\Microsoft.Psi.RealSense_Interop.Windows.x64.vcxproj" />
		<ProjectReference Include="..\Sources\Runtime\Microsoft.Psi.Interop\Microsoft.Psi.Interop.csproj" />
		<ProjectReference Include="..\Sources\Runtime\Microsoft.Psi.Windows\Microsoft.Psi.Windows.csproj" />
		<ProjectReference Include="..\Sources\Runtime\Microsoft.Psi\Microsoft.Psi.csproj" />
	</ItemGroup>

</Project>
```

Here is my code:
```
using System;
using Microsoft.Psi;
using Microsoft.Psi.RealSense.Windows;

namespace ConsoleApp1
{
    class Program
    {
        static void Main(string[] args)
        {
            var p = Pipeline.Create();
            RealSenseSensor r = new(p);
            r.ColorImage.Do(t => Console.WriteLine(t));
            p.RunAsync();
            Console.ReadKey();
            p.Dispose();
        }
    }
}
```

I have attempted a number of fixes as seen near the end of this discussion: https://github.com/microsoft/psi/discussions/167

Thank you.

chitsaw, 2021-06-15:
Could you verify that the path to the RealSense binaries is in your system PATH variable? Typically this would be something like _"C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64"_, depending on where you installed the RealSense SDK.

trigaten, 2021-06-15:
The `RealSenseSDKDir` PATH variable is set to `C:\Program Files (x86)\Intel RealSense SDK 2.0` per the Wiki. 
Here are the files at that location:
[image]

chitsaw, 2021-06-16:
The `RealSenseSDKDir` environment variable is only required in order to build, which you clearly have been able to do. However, your issue occurs at runtime, and the exception message suggests that a dependent DLL failed to load. That's what prompted me to ask that you check your system's `Path` environment variable.

Since the \psi RealSense component depends on the Intel RealSense components at runtime, the location of those files (_C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64_) needs to be in the system path.

If you don't see the above path (including the _...\bin\x64_ part) in your `Path`, I would try adding it, restarting Visual Studio (in order for it to pick up the updated path), then running your app again.

trigaten, 2021-06-16:
Ah, sorry, I misunderstood. Adding that path to my PATH has resolved this issue. 

Thank you very much.

closed 2021-06-16 by trigaten