microsoft/psi #186: Question: How to configure the project in visual studio 2019 to make it run on GPU?

Shawn1222, 2021-08-04:
Hello,
I am trying the "WhatIsThat" sample and wondering how to make it run on GPU. My GPU is Nvidia RTX 3080 Ti and I have installed and configured CUDA v11.4 and cuDNN on Windows 10. I also set the "CpuOnlyMode" to false in the AzureKinectBodyTrackerConfiguration. Now I am wondering how to configure in visual studio 2019 to make the project run on GPU. Can you give me any instructions?

Looking forward to your reply. Thanks.

xiangzhi, 2021-08-04:
Are you able to get the application running at all?

I know the Nuget version of the Azure Kinect SDK used in Psi is an older version and does not work with RTX 30XX cards (#116).

Shawn1222, 2021-08-05:
Yes, I can run the sample but no body detection results showed on \psi studio.

Besides, I have tried to replace related .dll files such as "cudnn64_7.dll" with "cudnn64_8.dll" in CUDA v11.4. However, I got the run error that fails to find the GPU device. 

Is there any other solution you suggest?

Many thanks.

xiangzhi, 2021-08-05:
Do you mind checking the version of Azure Kinect SDK and Body Tracking SDK you have installed on your system? [Does the body tracking viewer that comes with the SDK work for you?](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-setup#verify-body-tracking)

I think the GPU problem lies in Psi using Azure Kinect Body Tracking version 1.0.0. [Microsoft only added support for RTX 30XX card in Azure Kinect Body Tracking version 1.1.0.](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download#v110). If that is the cause of your problem, our lab has a branch on our repository that uses the latest version of the SDK with some tweaks but it would require you change the sample to use it. (link:[https://github.com/CMU-TBD/psi/tree/dev/k4abt-1.1](https://github.com/CMU-TBD/psi/tree/dev/k4abt-1.1)) Unfortunately, I don't have a RTX 30XX card with me to test the sample.

Also, out of curiosity, does the sample work in CPU only mode?

Hope this help!!

Shawn1222, 2021-08-09:
Hi,
The version of Azure Kinect SDK is v1.4.1 and Body Tracking SDK version is 1.1.0 on my Win10. The body tracking viewer works and can show the body detection results.

I agree with you that the problem is caused by the Azure Kinect Body Tracking v1.0.0 which does not support RTX 30XX. However, the latest version 1.1.0 does not release the Nuget package. I will have a try with your lab branch.

The sample works in CPU only mode and my CPU is intel i9-10900K.

Thanks for help.