microsoft/psi #193: Qustion about the function Interpolate in NearestReproducibleInterpolator

HaipingLe, 2021-08-16:
Code link:
https://github.com/microsoft/psi/blob/master/Sources/Runtime/Microsoft.Psi/Common/Interpolators/NearestReproducibleInterpolator.cs#L85

Question Description:
If I specify the timespan in the interpolater with （TimeSpan.FromSeconds(-2), TimeSpan.FromSecond(2)), it means that  i want to find a nearest message from the secondary stream with in the time window[-2,2] ( maybe (-2,2) ).

In thr foreach body,
=================================================================
  // We stop searching either when we reach a message that is beyond the window end
                    // when the distance (absolute delta) exceeds the minimum distance.
                    if (messageIsOutsideWindowEnd || (distance > minDistance))
                    {
                        break;
                    }

                    // keep track of the nearest match so far and its delta
                    nearestMatch = message;
                    minDistance = distance;
                }

                i++;
==================================================================

For example, the first message from secondaryQueue can be reagrd as the first one nearst message, the distance will be updated the new distance. Then we pick the second message from secondaryQueue, if the new distance is bigger than the minDistance  updated by the first message, it will stop searching.   Is my understanding correct? 

I know it cost time to get my point, i 'd like to say thank you very much if you can read it carefully.

sandrist, 2021-08-16:
Yes, your understanding is correct. In case it's helpful, we also have some helpful explanations and further documentation about interpolators under the wiki page on [Stream Fusion and Merging](https://github.com/microsoft/psi/wiki/Stream-Fusion-and-Merging)

HaipingLe, 2021-08-17:
@sandrist ，
Thanks for your response.
I think the method of finding  the nearest message in this case works well.
In general, the messages in a message queue  are orded by the originating time so that this algorithm works.

closed 2021-08-18 by HaipingLe