microsoft/psi #203: Help with debugging exception when openning/reading from PsiStore

xiangzhi, 2021-09-10:
We have been collecting our study data using Psi and ran into a problem with one particular store which can be opened in PsiStudio but failed to be readable by code. An example of the code is as following:
```csharp
            var datasetPath = @"E:\Processed-Study-Data\p127\p127.pds";
            var dataset = Dataset.Load(datasetPath);
            var tourSession = dataset.Sessions.Where(m => m.Name.Contains("tour")).First();
            using var p = Pipeline.Create();
            var importer = PsiStore.Open(p, tourSession.Partitions[0].Name, tourSession.Partitions[0].StorePath);
            var studyStateStream = importer.OpenStream<Tuple<string, TimeSpan>>("study.state");
            p.Run(ReplayDescriptor.ReplayAll);
``` 

When we try to open/read from stream using code. It throws errors like the following:
```
System.IO.FileNotFoundException
  HResult=0x80070002
  Message=Unable to find the specified file.
  Source=System.Core
  StackTrace:
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.MemoryMappedFiles.MemoryMappedFile.OpenCore(String mapName, HandleInheritability inheritability, Int32 desiredAccessRights, Boolean createOrOpen)
   at System.IO.MemoryMappedFiles.MemoryMappedFile.OpenExisting(String mapName, MemoryMappedFileRights desiredAccessRights, HandleInheritability inheritability)
   at Microsoft.Psi.Persistence.InfiniteFileReader.LoadNextExtent() in C:\Users\Zhi\source\repos\CMU-TBD\psi\Sources\Runtime\Microsoft.Psi\Persistence\InfiniteFileReader.cs:line 240
```
It states the file is `"recording.Data_320430741.psi"`. Opening other streams would throw exceptions such as:
```
System.AccessViolationException
  HResult=0x80004003
  Message=Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
  Source=Microsoft.Psi
  StackTrace:
   at Microsoft.Psi.Persistence.InfiniteFileReader.MoveNext() in C:\Users\Zhi\source\repos\CMU-TBD\psi\Sources\Runtime\Microsoft.Psi\Persistence\InfiniteFileReader.cs:line 124
```
```
System.AccessViolationException
  HResult=0x80004003
  Message=Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
  Source=Microsoft.Psi
  StackTrace:
   at Microsoft.Psi.Persistence.InfiniteFileReader.MoveNext() in C:\Users\Zhi\source\repos\CMU-TBD\psi\Sources\Runtime\Microsoft.Psi\Persistence\InfiniteFileReader.cs:line 124
```

Do you all have any tips to salvage this psi store or ways to debug the issue? Unfortunately, I cannot share the PsiStore for debugging due to PII issues and IRB.

chitsaw, 2021-09-10:
Interesting. From the call stack you provided it looks as though the `InfiniteFileReader` thinks the store path is null. I would put a breakpoint in your code and check the value of `tourSession.Partitions[0].StorePath` in the first instance. If it is null, that might point to some problem with loading/interpreting the .pds file. I understand you are not able to share the store due to data privacy reasons, but perhaps you could share the .pds file if you are still unable to resolve this after further debugging.

xiangzhi, 2021-09-10:
Thanks for the quick response! I just checked the name and store path, both are pointing to the correct location (name:`recording`, storePath: `E:\\Processed-Study-Data\\p127\\recordings\\p127.tour.0\\recording`). I also included the pds file below.

One mysterious thing about this store is that I was able to run `CreateDerivedPartitionAsync` on it to create the `2DRecentered` partition. It only fails when I try to open streams programmatically. 
```
{
  "Name": "p127",
  "Sessions": [
    {
      "Name": "p127.tour.0",
      "Partitions": [
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.IStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "recording",
          "StoreName": "recording",
          "StorePath": "recordings\\p127.tour.0\\recording",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        },
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "2DRecentered",
          "StoreName": "2DRecentered",
          "StorePath": "recordings\\p127.tour.0\\2DRecentered",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        }
      ]
    },
    {
      "Name": "p127.checkin.0",
      "Partitions": [
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.IStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "recording",
          "StoreName": "recording",
          "StorePath": "recordings\\p127.checkin.0\\recording",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        },
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "2DRecentered",
          "StoreName": "2DRecentered",
          "StorePath": "recordings\\p127.checkin.0\\2DRecentered",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        }
      ]
    },
    {
      "Name": "p127.handoff.0",
      "Partitions": [
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.IStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "recording",
          "StoreName": "recording",
          "StorePath": "recordings\\p127.handoff.0\\recording",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        },
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "2DRecentered",
          "StoreName": "2DRecentered",
          "StorePath": "recordings\\p127.handoff.0\\2DRecentered",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        }
      ]
    },
    {
      "Name": "p127.failure.0",
      "Partitions": [
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.IStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "recording",
[... 29 lines cut ...]
        }
      ]
    },
    {
      "Name": "p127.delivery.0",
      "Partitions": [
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.IStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "recording",
          "StoreName": "recording",
          "StorePath": "recordings\\p127.delivery.0\\recording",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        },
        {
          "$type": "Microsoft.Psi.Data.Partition`1[[Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi]], Microsoft.Psi.Data",
          "Name": "2DRecentered",
          "StoreName": "2DRecentered",
          "StorePath": "recordings\\p127.delivery.0\\2DRecentered",
          "StreamReaderTypeName": "Microsoft.Psi.Data.PsiStoreStreamReader, Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null"
        }
      ]
    }
  ]
}
```

xiangzhi, 2021-09-10:
I simplified the code to this and the exception still occurs:
```csharp
        public static void Run()
        {
            using var p = Pipeline.Create();
            var importer = PsiStore.Open(p, "recording", @"E:\Processed-Study-Data\p127\recordings\p127.tour.0\recording");
            var studyStateStream = importer.OpenStream<Tuple<string, TimeSpan>>("study.state").Select(m => m.Item1);
            p.Run(ReplayDescriptor.ReplayAll);
        }
```

chitsaw, 2021-09-10:
I assume that the folder "E:\Processed-Study-Data\p127\recordings\p127.tour.0\recording" exists and that it contains files such as "recording.Catalog_000000.psi", "recording.Data_000000.psi", etc.?

When I tried replicating your .pds file using the exact same folder structure with my own store store, and I am able to open streams without exception. I just re-read your initial post and you mentioned that the file name was "recording.Data_320430741.psi". Does that file exists? Seems a little odd as that would imply your store contains a ton of data (maybe it does)? Does the exception happen right away or does it occur after running for some time?

xiangzhi, 2021-09-10:
You are correct. I don't have that much data. Here's a screenshot of that particular store:
[image]

The exception happened rather quickly, Here's a video of me stepping through the callstack right before the exception is thrown. The video ends with the function that throws the error.
https://user-images.githubusercontent.com/3656500/132926077-a4de25f1-211f-4a1e-a649-c63fd90f004d.mp4

chitsaw, 2021-09-10:
Thanks for the video, that helps. Will investigate more to see if we can repro this.

xiangzhi, 2021-09-10:
Thanks! I know it is not much to build off from. Let me know if there is any particular things you would like me to check. It only happened to 1 out of 330 stores.

chitsaw, 2021-09-13:
Unfortunately I have not been able to repo this issue.

From your video, I noticed that `this.remainingBlockSize < 0` as soon as you hit the first breakpoint in `InfiniteFileReader.MoveNext()`. I would have expected `remainingBlockSize` to be a positive value (it represents the length of the next message in the data store). Only when we read to the end of the file extent (e.g. recording.Data_000000.psi) should we expect this to be a negative value, representing the next file extent's id as a negative number (the negative value is how we tell we've reached the end of the file). The fact that we're hitting this condition right away, with the strange value of -320430741 indicates that there may be some sort of data error/corruption in the first extent file recording.Data_000000.psi.

However, it is strange that you were able to run `CreateDerivedPartitionAsync` successfully on the store. If the store was indeed corrupted, I would have expected that to fail also as it goes through the same file reader code path as your code that opens the store programmatically.

At this point, I'm afraid that all I can suggest is maybe stepping through the `MoveNext()` method both in the `CreateDerivedPartitionAsync` case as well as the failure case to see how they differ. You may also be able to use `CreateDerivedPartitionAsync` to create a duplicate of the original store and salvage the data that way.

Alternatively, you may also want to examine your data file in a binary editor to see if you can spot anything that's out of place with the first message entry (since you seem to be hitting the error right away). The format of the psi data file is a sequence of encoded messages as follows:
```
MessageLength: 4 bytes
Envelope: 24 bytes
    SourceId: 4 bytes
    SequenceId: 4 bytes
    OriginatingTime: 8 bytes
    CreationTime: 8 bytes
MessagePayload: (MessageLength - 24) bytes
...
```
The sequence of messages ends with a negative 32-bit signed integer representing the next file id (e.g. -1, -2, -3, etc.). The file id is the number after the ".Data_" in the file name.

Hope this helps.

xiangzhi, 2021-09-14:
Thanks for looking into it! I suspected it is a very weird edge case in my data.

I'll step through the code and see whether I can spot the problem and report back. The new comments you added also looks helpful!

xiangzhi, 2021-09-14:
I did a bit more digging and found the issue is in `PageIndexCache` and the `Index` file. The index it found for the given time was invalid for the `data_*`files. The position was pointing to a middle of a message, hence returning a negative value.

The reason `CreateDerivedPartitionAsync` end up working because it opened a stream that has a way earlier originating time and had a different index entry. This is probably why PsiStudio worked too because it read from the very beginning. 

For reference:
The `CreateDerivedPartitionAsync` started with index 0 with the following IndexEntry
```
{Microsoft.Psi.Data.IndexEntry}
    CreationTime: Hour = 0
    ExtentId: 0
    OriginatingTime: Hour = 0
    Position: 0
```

The example code started with index 8 with the following IndexEntry
```
{Microsoft.Psi.Data.IndexEntry}
    CreationTime: Hour = 16
    ExtentId: 0
    OriginatingTime: Hour = 16
    Position: 1392832
```
I checked the time in the IndexEntry and it looks valid. So, I'm not sure what I did that got the index file corrupted. I tried recreating the error but the new files have a valid index file. This might be a mystery that we can never solve since even I cannot recreate it. 

Since there is no actual data, I included the index files if y'all want to take a look and if it might give you a clue on what might have happened.
[Archive.zip](https://github.com/microsoft/psi/files/7164725/Archive.zip)

chitsaw, 2021-09-15:
Yeah, I'm not sure how easy it would be to get to the bottom of this by forensic analysis. The index entries in the index file you provided look fine, at least up to the offending entry. Since you say that position points to the middle of a message, then it could be that it is the data file for extent 0 that has been corrupted, with extra/missing bytes in the region prior to position 1392832.

xiangzhi, 2021-09-17:
I agree, it's going to be hard to figure it out. It's possible it was some weird circumstances when I ran the code.  I was able to retrieve the data, so there isn't anything blocking my research now. Thank you so much for looking into this.

closed 2021-09-17 by xiangzhi