microsoft/psi #208: Allow to overwrite ONNX Model Shape

ankitbko, 2021-11-28:
I was unable to run DB_TD500_resnet50.onnx model (https://docs.opencv.org/4.x/d4/d43/tutorial_dnn_text_spotting.html) using `OnnxModel` as it was throwing following error - `Length of memory (1625088) must match product of dimensions (3)`.

The only way to fix it was to duplicate `OnnxModel` locally and pass the correct `shapeDictionary` when applying ONNX Model

```csharp
this.context.Transforms.ApplyOnnxModel(
      modelFile: configuration.ModelFileName,
      outputColumnNames: new[] { configuration.OutputVectorName },
      inputColumnNames: new[] { configuration.InputVectorName },
      gpuDeviceId: configuration.GpuDeviceId,
      shapeDictionary: new Dictionary<string, int[]>()
          {
             { "input", new[] { 1, 3, 736, 736} }
          });
```

Request is to allow overriding `shapeDictionary` when creating `OnnxModel`. This overriding requirement also exist to run FasterRCNN model https://github.com/dotnet/machinelearning/pull/3963

ankitbko, 2021-11-29:
I am adding another ask to same issue as it deals with same config class - Allow multiple input and output column names to be passed to `OnnxModel`. The `ApplyOnnxModel` accepts list of output and input names so `OnnxModelConfiguration` should also accept a list along with shape data for each.

danbohus, 2021-12-06:
Thanks for flagging this! Agreed, the `OnnxModel` and `OnnxModelConfiguration` classes should be updated to allow for configuring lists of input/output names, and to enable the shapeDictionary parameter. The change seems relatively straightforward, and we would welcome a PR on this. Let us know if you plan to make this contribution. Otherwise we'll try to get to it in the near future.

ankitbko, 2021-12-09:
Sure will be happy to contribute.