microsoft/psi #210: Different bitmap output based on makeCopy parameter in ToBitmap method

ankitbko, 2021-12-09:
```csharp
public override async void Receive(Message<Shared<Image>> message)
{
    var bmp = message.Data.Resource.ToBitmap(false);
    bmp.Save("img.jpg", System.Drawing.Imaging.ImageFormat.Jpeg);

    // ...... removed for brevity
}
```
When `var bmp = message.Data.Resource.ToBitmap(false);` I get this image
[image]

When I change the Boolean `using var bmp = message.Data.Resource.ToBitmap(true);` I get this image.
[image]

Not sure if this is a bug or not.

sandrist, 2021-12-11:
Hmm, I wasn't able to replicate this behavior with a simple test. I saved your second image as "C:\Temp\test-image.png", loaded it as a \psi `Image`, and called `ToBitmap` in both ways.

```csharp
var image = Image.FromBitmap(new Bitmap(@"C:\Temp\test-image.png"));
image.ToBitmap(false).Save(@"C:\Temp\test-image-false.jpg", System.Drawing.Imaging.ImageFormat.Jpeg);
image.ToBitmap(true).Save(@"C:\Temp\test-image-true.jpg", System.Drawing.Imaging.ImageFormat.Jpeg);
```

but both resulting images look fine...

There still might be a bug in `ToBitmap` if something is screwed up with stride or pixel format, but I think I would have to see more of your code showing how the `Shared<Image>` stream gets created that is being passed to your `Receive` method.

ankitbko, 2021-12-11:
After troubleshooting together, we found out that the incorrect behavior happens in linux only and not in windows (mac is untested). The sample code that @sandrist wrote leads to reproducible behavior in linux machine.