microsoft/psi #213: Opening Image stream throws error

Bongo9911, 2022-01-31:
I'm attempting to open an Image stream like shown in the [Kinect example](https://github.com/microsoft/psi/wiki/Azure-Kinect-Overview) like so:

```
using (var pipeline = Pipeline.Create())
{
    var data = PsiStore.Open(pipeline, store, Path.GetFullPath(path));
    var images = data.OpenStream<Shared<Image>>(stream);
    pipeline.Run();
}
```

With path referring to the location of my psi files, and the stream being the Image stream of the store.

However, I'm running into this error:

**System.Runtime.Serialization.SerializationException: 'The type Microsoft.Psi.Shared`1[[Microsoft.Psi.Imaging.Image, Microsoft.Psi.Imaging, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null]], Microsoft.Psi, Version=0.15.49.1, Culture=neutral, PublicKeyToken=null changed between versions from struct to class, which is not supported.'**

This detailed exception will only appear when I set the value of the stream variable to something other than an image stream though. Otherwise, I just get this generic error:

**System.AggregateException: 'Pipeline 'default' was terminated because of one or more unexpected errors'**
**NullReferenceException: Object reference not set to an instance of an object.**

Any ideas on how to address this? All the examples I've seen online and on the wiki don't seem to work for me.

danbohus, 2022-02-01:
I believe the first (serialization) exception you're seeing is because in that case the stream you're trying to open is not actually of type `Shared<Image>`. So that's actually not the right exception to be trying to debug, since I assume you want to actually open the image stream. 

You mentioned that when you open the correct (image) stream, you get an `AggregateException`. This happens because while you were running the pipeline, one of the components probably throws a `NullReferenceException`, and the psi runtime terminates the pipeline when a component throws an exception and reports this `AggregateException`. 

To find out what causes the original `NullReferenceException`, I think you have to setup Visual Studio to break when it encounters any Common Language Runtime (CLR) exceptions. You can do so by going to the Exception Settings Window (from the Debug menu -> Windows -> Exception Settings, and in that window make sure that the checkbox for "Common Language Runtime Exceptions" is checked (i.e. has a checkmark in it, not a filled square, and not empy). When you rerun your app, it should stop in the debugger at the moment and place where the `NullReferenceException` happens, and hopefully this will help you figure out the problem is.

Bongo9911, 2022-02-02:
I made sure the "Common Language Runtime Exceptions" checkbox was checked.

I may not have made my original post clear though, the null exception happens at the `pipeline.Run()` line, and it still gives the exception at that line after checking that checkbox.

I don't really understand how it could be an issue with the pipeline, so maybe it's an issue with my store file or the stream itself? I'm not really sure of a good way to test that though. I have another stream in the store that works fine with this pipeline (although replacing the `OpenStream()` with the appropriate type). Looking at the type of my Image stream, I get this output:

**Microsoft.Psi.Shared`1[[Microsoft.Psi.Imaging.EncodedImage, Microsoft.Psi.Imaging, Version=0.14.35.3, Culture=neutral, PublicKeyToken=null]], Microsoft.Psi, Version=0.14.35.3, Culture=neutral, PublicKeyToken=null**

I don't see any clear issues with anything thus far, but maybe you can diagnose something from this?

Bongo9911, 2022-02-05:
I figured it out, I needed to be using Shared<EncodedImage> instead of Shared<Image>

closed 2022-02-05 by Bongo9911