microsoft/psi #220: Renaming Streams in PsiStudio

NathanielKBlan, 2022-04-08:
Hi, I was wondering if there is any way to rename streams within PsiStudio.

sandrist, 2022-04-08:
There is not currently a way to rename streams in place, without writing a new store. And there is not an easy way to achieve this in PsiStudio, without writing some custom code.

The easiest path might be to write a little app that simply [loads the store](https://github.com/microsoft/psi/wiki/Brief-Introduction#4-replaying-data) and persists all the streams [to a new store](https://github.com/microsoft/psi/wiki/Brief-Introduction#3-saving-data), changing any names that you wish along the way. You can then load the new store in PsiStudio.

Another potential path would be to define a new [BatchProcessingTask](https://github.com/microsoft/psi/blob/master/Sources/Data/Microsoft.Psi.Data/BatchProcessingTask.cs) that rewrites stream names as you wish. You could then [point PsiStudio to the assembly containing that task](https://github.com/microsoft/psi/wiki/3rd-Party-Visualizers), and then execute the task on your data within PsiStudio. This approach isn't well documented, so let me know if you're interested in this path, and I can help guide.

sandrist, 2022-04-09:
Also, to preserve meaningful message times and latencies in the new streams, make sure to execute the pipeline with a `ReplayAllRealTime` [Replay Descriptor](https://github.com/microsoft/psi/wiki/Pipeline-Execution#predefined-replay-descriptors).

closed 2022-09-28 by sandrist