microsoft/psi #233: Can't visualize HoloLens2 streams in Psi studio.

mrcfschr, 2022-06-06:
#### Thank you for this amazing codebase! I'm completely new to using psi and just tried the HoloLens Capture app/server. It works as expected except I can't visualize the sensor streams( rgb, depth ir etc) in psi studio. I would really appreciate some advise on how to get this to work!

#### Steps to reproduce the issue 

1.  build psi from source https://github.com/microsoft/psi/wiki/Building-the-Codebase 
System Specs:
- Processor	Intel(R) Core(TM) i7-10700 CPU @ 2.90GHz   2.90 GHz
- Installed RAM	16.0 GB
- System type	64-bit operating system, x64-based processor
- Edition	Windows 10 Home
- Version	21H2
- OS build	19044.1706

2.  deploy /run HoloLensCapture to HL2 (research mode enabled, give permissions, change capture server ip to pc ip)
3.  run the server
[image]
4. open store in psi studio
[image]
5. no images in psi studio: run HL exporter after stopping app via stop button, this does get images from the same stores?
[image]

#### What's the expected result?

- being able to visualize the HL data streams in psi studio

#### What's the actual result?

- only shows frame rate/ latency info, eye gaze etc but no image streams

#### Additional details / screenshot
- running Test.Psi.PsiStudio gives this output:
[image]

sandrist, 2022-06-06:
Thanks for the detailed issue. There are a couple different ways to visualize the captured images (and we'll improve our documentation to make these more clear):
1. Note that the `VideoEncodedImageCameraView` stream has a data type of `EncodedImageCameraView`. The visualizer for this type is found in `Microsoft.Psi.Spatial.Euclidean.Visualization.Windows`. You'll need to build that project, and [point PsiStudio to the resulting dll](https://github.com/microsoft/psi/wiki/3rd-Party-Visualizers). Make sure to close PsiStudio and re-open it once you've edited the settings file. You should then see an option to visualize that stream as a 3D "image camera view", which draws the image with a frustum at the 3D pose of the camera.

2. You can also right-click on the stream and select "Expand Members" (or simply double-click on it). This will automatically create three "derived" streams which represent each class member: the `CameraIntrinsics`, `CameraPose`, and `ViewedObject` (in this case the viewed object is an image). You should be able to right-click on `ViewedObject` and select "Visualize as Image" to see the 2D image, without needing to load any additional 3rd party visualizer DLLs.

[image]

mrcfschr, 2022-06-06:
Thank you so much this works like a charm!!

closed 2022-06-06 by mrcfschr