microsoft/psi #257: FFmpeg probelm

HaozheQi, 2022-09-07:
Hello!

I want to use the [HoloLens Capture Exporter](https://github.com/microsoft/psi/tree/master/Sources/MixedReality/HoloLensCapture/HoloLensCaptureExporter) to export the data recorded by Hololens. It needs the FFMPEG Support to build the Microsoft.Psi.Media.Windows.x64.

However, when I include the recent version of ffmpeg into the project, it reports `'avcodec_register_all': identifier not found`. It seems that `avcodec_register_all( )` in this [code](https://github.com/microsoft/psi/blob/master/Sources/Media/Microsoft.Psi.Media.Native.x64/FFMPEGReaderNative.cpp#L208) has already been deprecated since ffmpeg 4.0. 

Based on this, I wonder if it is possible to share the `ffmpeg-20180227-fa0c9d6-win64-dev` folder to me that you are currently using, because it is hard to find the ffmpeg of old develop version?

sandrist, 2022-09-08:
Please look at my answer to your [other issue here](https://github.com/microsoft/psi/issues/258). In short, you shouldn't need FFMPEG in order to build these projects, as long as you haven't defined the `FFMPEGDir` environment variable on your machine.

chitsaw, 2022-12-09:
Resolved 706ae03ba548a4e5696d959050d6476828dde979

closed 2022-12-09 by chitsaw