microsoft/psi #262: Will the implementation of "Continual Learning about Objects in the Wild: An Interactive Approach" be opensource?

Abdul-Mukit, 2022-10-27:
Hello,
I am studying the article "Continual Learning about Objects in the Wild: An Interactive Approach". I was wondering if the implementation of this system will be available for us to experiment with.
How can one contribute to this research? The project aligns very much with my interest in AR + CV. I was wondering if there is a way to contribute. Psi + Mixed Reality, is truly a powerful tool. Thank you.

sandrist, 2022-10-28:
I'm glad to hear that you found [this paper](http://erichorvitz.com/Continual_learning_ICMI_2022.pdf) to be interesting! The specific system described in that paper is quite complex, and we're not sure if we'll be able to release the full implementation in the short term. However, we do plan to keep building and releasing new features and components for \\psi + mixed reality going forward, and we hope it will continue to become easier to use these components and tools to build similar end-to-end applications for your own scenarios. Stay tuned to the repo for more updates soon.

In the meantime, feel free to contribute to this open source project in whatever way you'd like, either through new components, new samples, bug fixes, etc. Take a look through some of the existing issues and discussion topics for possible ideas. We have posted, and will continue to post, various "Help Wanted" topics that we feel might be particularly good opportunities for contribution. One large area of potential contribution would be to create new \\psi components that wrap some of the latest and greatest AI models out there. Language models, vision models, multimodal models, etc. It seems like more of these models are released every day, many of them open source, and the community could benefit from easy ways to use these models and integrate with larger systems (including but not limited to mixed reality scenarios), along with samples of what that might look like. We would be happy to accept contributions of that form, or point to external repositories that build such components and samples on top of \\psi.

Abdul-Mukit, 2022-10-28:
Got it. Thank you for your reply.

closed 2022-10-28 by Abdul-Mukit