microsoft/psi #277: HoloLens 2 ir cam intrinsics

mrcfschr, 2023-03-10:
I'm trying to track a surgical tool using the Hololens' IR sensor and PSI. Since the psi intrinsics are only estimates, I was wondering if someone in the community knows a better way to get more accurate intrinsics. As it is right now, we have a bit of an offset (a couple of mm) when tracking the tool.
Thank you!

sandrist, 2023-03-11:
Yes, maybe experiment with the MapImagePointToCameraUnitPlane function directly and see if that gives you better/different results? Beyond that, take a deeper look at the [Research Mode](https://github.com/microsoft/HoloLens2ForCV) APIs to see if there is anything there. We use Research Mode libraries to expose the IR sensor to \psi, but there might be more there we could leverage for computing intrinsics. Not sure though.

Abdul-Mukit, 2023-03-13:
@mrcfschr hi,
As far as I remember, I had performed camera calibration using a board. https://docs.opencv.org/4.x/dc/dbb/tutorial_py_calibration.html

I captured a couple of images from the IR sensor. They were in 16 bits so I had to convert them to 8 bits. OpenCV's histogram equalization function didn't support an input of 16 bits. Then I performed a histogram equalization so that the corner points are detectable in the camera calibration OpenCV functions.

mrcfschr, 2023-03-14:
Thank you I'll give it a try!

mrcfschr, 2023-03-16:
I set the Transform of the CameraIntrinsics to the opencv cameramatrix and the tangential distortion to the 3rd and 4th opencv distortion coefficient and the radial distortion to the 1st 2nd and 5th Opencv distortion coefficient based on  dist = (k1,k2,p1,p2[,k3[,k4,k5,k6[,s1,s2,s3,s4[,τx,τy]]]])https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga687a1ab946686f0d85ae0363b5af1d7b.
Am I setting this right?  Because I'm getting more of an offset now, may also be due to something else in the code will use aruco for separate validation:
```
 //            calibrated using checkerboard
                /*total error: 0.04400390079043265
                Intrinsic parameters(camera matrix):
                [[245.31616967   0.         255.09594219]
                 [  0.         246.12117205 247.46196344]
                 [  0.           0.           1.        ]]
                Distortion coefficients: (k1,k2,p1,p2[,k3[,k4,k5,k6[,s1,s2,s3,s4[,τx,τy]]]])  https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga687a1ab946686f0d85ae0363b5af1d7b
                [[-3.00205711e-01  9.69518716e-02  3.91632399e-03 - 8.20810782e-05
                  - 1.39891582e-02]]
                newcameramtx:
                                [[141.71080017   0.         251.40767011]
                 [  0.         143.26325989 250.54294455]
                 [  0.           0.           1.        ]]*/
                ICameraIntrinsics camInt = camInfo.Item1.CameraIntrinsics;
                var values = new double[,]
                {
                    { 245.31616967, 0.0, 255.09594219 },
                    { 0.0, 246.12117205, 247.46196344 },
                    { 0.0, 0.0, 1.0 }
                };
                MathNet.Numerics.LinearAlgebra.Matrix<double> customTrans = MathNet.Numerics.LinearAlgebra.Matrix<double>.Build.DenseOfArray(values);
                var valuesTDist = new double[] { 3.91632399e-03, -8.20810782e-05 };
                MathNet.Numerics.LinearAlgebra.Vector<double> customTangentialDistortion = MathNet.Numerics.LinearAlgebra.Vector<double>.Build.DenseOfArray(valuesTDist);
                var valuesRDist = new double[] { -3.00205711e-01, 9.69518716e-02, -1.39891582e-02, 0.0, 0.0, 0.0 };
                MathNet.Numerics.LinearAlgebra.Vector<double> customRadialDistortion = MathNet.Numerics.LinearAlgebra.Vector<double>.Build.DenseOfArray(valuesRDist);
                camInt = new CameraIntrinsics(camInt.ImageWidth, camInt.ImageHeight, customTrans, customRadialDistortion, customTangentialDistortion, camInt.ClosedFormDistorts);
```

mrcfschr, 2023-03-29:
The tracked tool does look slightly more aligned but still not perfect.. closing this for now.

closed 2023-03-29 by mrcfschr