microsoft/psi #282: [Bug]: Hitting an exception when attempting to start hololenscaptureapp in

joshbobillot, 2023-05-23:
I'm hitting an issue when attempting to Launch the hololensCaptureApp after deploying to a hololens 2. VS hits an exception, and the app fails to begin capturing. The below exception is what I'm hitting.
[image]

Line 39 shows this.sensorDevice
[image]

and VS gives a suggestion to remove 'this' (not sure if this is the actual issue)
[image]

Repro steps:
Rebuild after cloning repository from here: https://github.com/microsoft/psi
Hololens connected to PC via USB cable
Add IP address from PC to CaptureServerIP.txt doc and add to hololens documents folder via device portal
Enable CaptureServer firewall settings to be both "Public" and "Private"
Deploy hololensCaptureApp to hololens 2
Start CaptureServer on PC
Start CaptureApp on HL2
click start from inside the app
Yellow square appears momentarily
Hololens freezes with black screen
Observe exception in VS

If I click on continue from the exception popup, the app re-appears in the hololens and just sits there saying "please wait constructing capture pipeline" and never progresses further.

Attempted on 2 machines, 1 laptop surface book and 1 desktop PC got same results.

Not sure if this is a build issue or if something is not configured correctly.

sandrist, 2023-05-23:
I think the VS suggestion is a red herring, but it looks like there is some exception in the initialization of the ResearchMode IMU component for some reason. Did you enable [Research Mode](https://learn.microsoft.com/en-us/windows/mixed-reality/develop/advanced-concepts/research-mode#enabling-research-mode-hololens-1st-gen-and-hololens-2) on your HoloLens device? (You may need to restart it once after enabling it). Other prerequisite steps can be found here: https://github.com/microsoft/psi/wiki/Mixed-Reality-Overview

joshbobillot, 2023-05-23:
After some more investigation I did end up putting it in research mode, thought it was fixed as after reboot was able to start the capture app without the exception, but now the app is just sitting there saying "Please wait Connecting to capture server <IP address>" but does not ever continue on past this state. I had also already installed the prerequisities from the page you linked too. Also on that note, I believe some of the 143 components are needed as I had to install those in order to clear up build errors I was getting earlier in the process.... these are what I have installed.... 

[image]

joshbobillot, 2023-05-23:
Only thing I am seeing now that is related to this comment in the prereques.... I do not recall being prompted for this 142 windows platform tools, but I assume 143 would have taken care of this, Could not having the 142 tools be causing my capture app to hang on connecting to server?
[image]

joshbobillot, 2023-05-23:
Well I installed C++ (142)UWPT, but capture app and sever still exhibiting the same behavior not seeming to be able to connect to each other. will have to try some more tomorrow.

sandrist, 2023-05-23:
It's hard to say what might be going wrong, but here are a few things you can try:
1. Uninstall the capture app from your HoloLens and deploy it from scratch
2. Check your firewall settings on Windows to ensure that the server application has both "Public" and "Private" set
3. Make sure the `CaptureServerIP.txt` file is present in your device's `User Folders\Documents` folder, and that it contains the actual IP address of the machine running the server
4. Run both the app and the server from Visual Studio in debug mode with all "Exception" settings turned on, to make sure there aren't any hidden exceptions being thrown that we're not seeing.

Let us know if any of these steps reveals anything.