microsoft/psi #296: PsiStoreReader has an inconsistent output for indexed entries on different projects (DotNet vs. UWP) same x64 architecture

austinbhale, 2023-10-11:
When there are indexed entries (i.e., enabling `largeMessages` for a stream writer), the `PsiStoreReader` does not parse the `IndexEntry` metadata correctly. I noticed the unexpected behavior occurs at the following line: https://github.com/microsoft/psi/blob/ef4b2a627ffefc5415b826930a883b309c0a37cd/Sources/Runtime/Microsoft.Psi/Persistence/PsiStoreReader.cs#L363

Expected (Write/Read on same project type):
[image]

Current (Write on UWP/DotNet, Read on DotNet/UWP):
[image]

## To Repro
1. Write to a \psi store with `largeMessages=true`
2. Read from a \psi store on the opposite platform (e.g., Write on .NET, Read on UWP & vice versa)

### UWP
```
namespace PsiImporterLargeMessageIssue
{
    using Microsoft.Psi;
    using Microsoft.Psi.Data;
    using System;
    using System.IO;

    internal class Program
    {
        private static void Main(string[] _)
        {
            const string StoreName = "Test";
            string StorePath = Windows.Storage.ApplicationData.Current.LocalFolder.Path;
            System.Diagnostics.Debug.WriteLine($"Store is located at: {Path.Combine(StorePath, StoreName)}");

            using (var pipeline = Pipeline.Create(StoreName))
            {
                PsiExporter writer = PsiStore.Create(pipeline, StoreName, StorePath, true);
                Generators.Repeat(pipeline, 1, 10, TimeSpan.FromMilliseconds(300)).Write("Test", writer, true);
                pipeline.Run();
            }

            using (var pipeline = Pipeline.Create(StoreName))
            {
                var reader = new PsiImporter(pipeline, StoreName, StorePath, true);
                reader.OpenStream<int>("Test").Do((msg, env) => System.Diagnostics.Debug.WriteLine($"Test: {msg}, {env.OriginatingTime.Ticks}"));
                pipeline.Run(ReplayDescriptor.ReplayAll);
            }
        }
    }
}
```

### DotNet
```
using Microsoft.Psi;
using Microsoft.Psi.Data;

const string StoreName = "Test";
const string StorePath = ""; // Check the debug output for UWP's ApplicationData path

// Here you can test the opposite direction: creating a store on DotNet and play it back on UWP
/*
using (var pipeline = Pipeline.Create(StoreName))
{
    PsiExporter writer = PsiStore.Create(pipeline, StoreName, StorePath, true);
    Generators.Repeat(pipeline, 1, 10, TimeSpan.FromMilliseconds(300)).Write("Test", writer, true);
    pipeline.Run();
}
*/

if (!PsiStore.Exists(StoreName, StorePath))
{
    throw new DirectoryNotFoundException();
}

using (var pipeline = Pipeline.Create(StoreName))
{
    var reader = new PsiImporter(pipeline, StoreName, StorePath, true);
    reader.OpenStream<int>("Test").Do((msg, env) => System.Diagnostics.Debug.WriteLine($"Test: {msg}, {env.OriginatingTime.Ticks}"));
    pipeline.Run(ReplayDescriptor.ReplayAll);
}
```

chitsaw, 2023-10-11:
Thanks for reporting this. We recently encountered this issue as well and are working on a fix. Are you by any chance running into this while using .NET 7? From our observations/testing thus far, the issue doesn't seem to occur on .NET 6 and below.

austinbhale, 2023-10-11:
That's awesome, glad y'all already saw this too! Yes, I was running it using .NET 7, falling back to .NET 6 is a great workaround :)

PR #297 (2023-10-20): Make struct layouts explicit
Fixes #296 where auto struct layouts have changed in .NET 7
closed 2023-10-20 by chitsaw