microsoft/psi #298: Could not load file or assembly 'Microsoft.Psi.Media_Interop.Windows.x64, Version=0.18.72.1, Culture=neutral, PublicKeyToken=null' or one of its dependencies.

yukw777, 2023-10-19:
When I try to run my project, I get the following error:

```
Unhandled Exception: System.IO.FileLoadException: Could not load file or assembly 'Microsoft.Psi.Media_Interop.Windows.x64, Version=0.18.72.1, Culture=neutral, PublicKeyToken=null' or one of its dependencies. 

Attempt to load an unverifiable executable with fixups (IAT with more than 2 sections or a TLS section.) (Exception from HRESULT: 0x80131019) ---> System.IO.FileLoadException: Could not load file or assembly 'Microsoft.Psi.Media_Interop.Windows.x64, Version=0.18.72.1, Culture=neutral, PublicKeyToken=null' or one of its dependencies. 

Attempt to load an unverifiable executable with fixups (IAT with more than 2 sections or a TLS section.) (Exception from HRESULT: 0x80131019) ---> System.IO.FileLoadException: Attempt to load an unverifiable executable with fixups (IAT with more than 2 sections or a TLS section.) (Exception from HRESULT: 0x80131019)
```

The workaround I've found is to include the following in my .csproj file:
```
    <Content Include="$(NuGetPackageRoot)/microsoft.psi.media.windows.x64/0.18.72.1-beta/lib/net472/Microsoft.Psi.Media_Interop.Windows.x64.dll">
      <CopyToOutputDirectory>Always</CopyToOutputDirectory>
    </Content>
```

Adding the following didn't work:
```
<CopyLocalLockFileAssemblies>true</CopyLocalLockFileAssemblies>
```

When I use `dotnet run` to run my project, it fails with the same error the first time, and it starts working again the second time. I'm assuming `dotnet run` somehow copies the dll over on the first run. Not sure why it fails every time on Visual Studio without the workaround above.

Any pointers would be appreciated!

sandrist, 2023-10-25:
Can you say a bit more about what kind of project you are trying to run? Is it a .NET framework project? Check that your .csproj is in the "new" SDK style: https://stackoverflow.com/questions/63055430/how-to-create-an-sdk-style-net-framework-project-in-vs

yukw777, 2023-10-27:
It’s a multi-framework project that targets net472 and net7.0. This error only happens for net472. Here’s a snippet of my csproj file (which I believe is already the new sdk style):

```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFrameworks>net472;net7.0</TargetFrameworks>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
    <PublishSingleFile>true</PublishSingleFile>
    <RuntimeIdentifiers>win-x64;linux-x64</RuntimeIdentifiers>
    <SelfContained>true</SelfContained>
    <Platforms>x64</Platforms>
  </PropertyGroup>
…

  <ItemGroup Condition="'$(TargetFramework)' == 'net472'">
    <Content Include="$(NuGetPackageRoot)/microsoft.psi.media.windows.x64/0.18.72.1-beta/lib/net472/Microsoft.Psi.Media_Interop.Windows.x64.dll">
      <CopyToOutputDirectory>Always</CopyToOutputDirectory>
    </Content>
  </ItemGroup>
…
```

Note that instead of using PublishSingleFile for net472, I use Costra.Fody.