microsoft/psi #301: PsiStudio not displaying correctly properties with new visualizers

AuMilliat, 2024-02-26:
Hello,

I'm working on new visualizers for PsiStudio, I have issue getting the property panel working as intended.
As an example, I've made a slight modification on the Azure Kinect Visualizer to display confidence as joint color (nothing changes in the properties).
 
[image]

Expected

[image]

From what I understand objects are not seen as expendable.

3D panel  

[image]

 I probably I'm doing wrong something and it will be fixed with #54

sandrist, 2024-02-26:
Can you share some of the new code?
Does `Skeleton` still have the `[ExpandableObject]` tag above it?

```csharp
/// <summary>
/// Gets the skeleton visualization object for the body.
/// </summary>
[ExpandableObject]
[DataMember]
[PropertyOrder(3)]
[DisplayName("Skeleton")]
[Description("The body's skeleton properties.")]
public SkeletonVisualizationObject<JointId> Skeleton { get; private set; }
```

AuMilliat, 2024-02-27:
Thanks for the reply,

Here the winmerge of the files (left: modified, right: base)
[image]

I've tried to copy-past the kinect visualization project outside \psi repo and I was able to reproduce the issue:

[image]

Integrating my classes inside the \psi repo correct the behavior:

[image]

So my guess it's related to dependencies.

AuMilliat, 2024-03-05:
I've corrected the issue by doing the link properly with \psi. I've build from the source those two nugets:
[image]

It might worth to publish them on nuget.org!

sandrist, 2024-03-05:
Thanks for the update! Glad that you are unblocked.
In the future we might put more visualizers in nugets and release PsiStudio as an installable executable. For now, we only support PsiStudio and related visualizers to be built from source.

closed 2024-03-05 by sandrist