microsoft/psi #308: Connecting windows C# and Ubuntu py with interop

sanmii, 2024-03-25:
Hi,
I am trying to publish some data from a windows machine and I would like to be able to read this data  from an ubuntu machine with python. I am doubting about how could I create this connection with the use of zmq.
The windows parto should be the set up this:
using (var p = Pipeline.Create())
{
    var mq = new NetMQWriter<double>(p, "tcp://localhost:12345", JsonFormat.Instance);
    var gen = Generators.Range(p, 0, 1000);
    var sin = gen.Select(x => Math.Sin(x / 100.0));
    var cos = gen.Select(x => Math.Cos(x / 100.0));
    sin.PipeTo(mq.AddTopic("sin-topic"));
    cos.PipeTo(mq.AddTopic("cos-topic"));
    p.Run();
}

An in ubuntu in order to be able to read this should be used something like this:
import zmq, json

socket = zmq.Context().socket(zmq.SUB)
socket.connect("tcp://windows_ip:12345")
socket.setsockopt(zmq.SUBSCRIBE, '') # '' means all topics

while True:
    [topic, message] = socket.recv_multipart()
    j = json.loads(message)
    print "Message: ", repr(j['message'])
    print "Originating Time: ", repr(j['originatingTime'])

I am mainly getting confused about which IP must be set in each place. 

I hope you can help me.

chitsaw, 2024-03-27:
From the above, it appears that all you would need to do would be to replace `windows_ip` in your python code with the actual IP address of your Windows machine and it should work. If this isn't the case, try replacing `localhost` in your Windows code with the same IP address of the Windows machine.

sanmii, 2024-04-07:
I made it work. I needed to use the IP of the local machine ( windows) for the publication of topic instead of localhost.

Thank you!

closed 2024-04-07 by sanmii