microsoft/psi #311: If operator to compare to ip producer variables in order to send a specific frame

sanmii, 2024-04-10:
Hey!
I am gathering data from two different cameras and I wnat to check which of the camera provides a higher value on confidence so I send the frame of this camera through the NetMQWriter. However I am quite stucked:

var decodedImageStream = imageStream.Decode();
var decodedImageStream2 = imageStream2.Decode();
var confidence = openFaceComponent.Conf;
confidence.Do(m => Console.WriteLine("conf 1 :" + m));
confidence.Write("Confidence", store);
var confidence2 = openFaceComponent2.Conf;
confidence2.Write("Confidence2", store);
confidence2.Do(m => Console.WriteLine("conf 2 :" + m));
var confidences = confidence.Join(confidence2, RelativeTimeInterval.Past());

So I want to check which of the confidences is higher and depending on that pipe to the NetQMWriter one decodedimage stream or the other. Can anybody help me with this?

sanmii, 2024-04-10:
I found a way! I joinned the confidences with the images and with a select I retuned the image with higher confidence and them I made a pipe to the NetMQWriter.

closed 2024-04-10 by sanmii

sandrist, 2024-04-10:
Glad you got it working!