microsoft/psi #313: Exporting audio and video from PSI store

kontogiorgos, 2024-05-03:
Hi,

Is there a way to get video and audio serialised in psi store into other formats (.wav, .mpeg, etc.)? I saw the cli tool to get data into csv or json, but I have not seen anything for audio/video data.

Thanks,
Dimos

danbohus, 2024-05-03:
Yes, you should be able to write a psi audio stream to a wav file using the [`WaveFileWriter`](https://github.com/microsoft/psi/blob/master/Sources/Audio/Microsoft.Psi.Audio/WaveFileWriter.cs) component, and also to write and audio and an image stream to an mpeg using the [`Mpeg4Writer`](https://github.com/microsoft/psi/blob/master/Sources/Media/Microsoft.Psi.Media.Windows.x64/Mpeg4Writer.cs) component. You'd have to write your own little utility tool that does this (i.e. reads the psi audio/image streams from the psi store and writes them to wav/mpeg using these components). Let us know if you run into trouble using these. I think you can find an actual example of exporting to wav and mpeg in the [HoloLensCaptureExporter](https://github.com/microsoft/psi/tree/master/Sources/MixedReality/HoloLensCapture/HoloLensCaptureExporter) app. 

As an aside, it would be great to add this functionality (i.e. of exporting an audio stream to wav or audio+images to mpeg) as an option in PsiStoreTool. If you decide to go that route, we'd welcome a PR on that front :).

kontogiorgos, 2024-05-03:
Got it, thank you Dan!

And yes, I think that PsiStoreTool functionality would be great.

sandrist, 2024-06-24:
Take a look for example at the example code snippet under [Data Replay](https://github.com/microsoft/psi/wiki/Pipeline-Execution#12-data-replay). This shows you how to read streams from a store. You could add some lines to this code that instantiate an Mpeg4Writer component, and pipe your streams into it.