microsoft/psi #315: sigma exception when connecting to server

codesForAnything, 2024-05-28:
[05/28/2024 16:01:09.8586]: SERVER PIPELINE RUNTIME EXCEPTION: One or more errors occurred.
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at Microsoft.Psi.Pipeline.<>c__DisplayClass110_0`1.<CreateAsyncReceiver>b__0(Message`1 m) in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Executive\Pipeline.cs:line 401
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass45_1`1.<TrackStateObjectOnContext>b__1() in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Executive\PipelineElement.cs:line 184
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass44_0.<TrackStateObjectOnContext>b__0() in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Executive\PipelineElement.cs:line 162
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass45_0`1.<TrackStateObjectOnContext>b__0(T m) in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Executive\PipelineElement.cs:line 184
   at Microsoft.Psi.Receiver`1.<>c__DisplayClass13_0.<.ctor>b__0(Message`1 m) in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Streams\Receiver{T}.cs:line 73
   at Microsoft.Psi.Receiver`1.DeliverNext() in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Streams\Receiver{T}.cs:line 347
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context) in D:\littleStar\github\psi\Sources\Runtime\Microsoft.Psi\Scheduling\Scheduler.cs:line 580
[05/28/2024 16:01:09.8616]:   Stopping Compute Server Pipeline @05/28/24 04:01:09 PM.
[05/28/2024 16:01:09.8636]:

closed 2024-05-29 by codesForAnything

codesForAnything, 2024-05-29:
and if you configure the azure openai key and endpoint, you should adjust the json file (https://github.com/microsoft/psi/blob/master/Applications/Sigma/Sigma/Resources/llmqueries.json) 
change the deployment model name into yours
[image]
eg, change **gpt-4** to **gpt-4o** for me

xyfhappy, 2024-11-13:
Hi, I have encountered some problems when run SIGMA. When I configure and run SIGMA according to “How to install, configure and run SIGMA”, I found that the server displays“RENDEZVOUS ERROR: Cannot find a recognizer with the required ID” in the terminal , at the same time, the initialization of the SIGMA APP on my Hololens2 has been stuck at 80%  for more than 30 minutes. 

Specifically, after I got Azure Speech Services, I set the appropriate environment variables based on my key and region, and configured a text file called “CognitiveServicesSpeechKey. txt”. I use the local anchors because I found that the Azure Spatial Anchors are no longer able to be registered. When I open SIGMA, I click START, then two options about anchors appear, no matter which one I click on, the initialization process ends up being stuck at 80%. The corresponding pictures are shown below, I would like to know what causes these phenomena.

Thank you for your time and consideration. I'm looking forward to your reply and any response is appreciated!

[image]
[image]
[image]