microsoft/psi #317: How to Determine if a Message is the Last Message?

KanaHayama, 2024-06-09:
Hi, I am writing a speech recognizer component that buffers audio and performs a one-time recognition of the buffered audio once the voice activity state becomes inactive. However, there is a problem when my audio is sourced from a wave file, where I cannot guarantee that the last message's voice activity property is inactive, thus the last buffered data cannot be posted to the downstream components.

I wonder if it is possible to know the final originating time of a `Receiver`. I know there is a `Pipeline.ComponentCompleted` event, but I guess I can't rely on it because my component does not know the name of the component which pipes to its `Receiver`, and there may be components with the same name. I also know I can get a final originating time if I implement `ISourceComponent`, but its semantic isn't what I want.

Could you tell me what is the best approach to know if a `Receiver` is receiving the last message, or to know the last message's originating time before the last message is received? Thank you.

danbohus, 2024-06-10:
You should be able to attach a handler to a receiver's `Unsubscribed` event. So for instance, from the constructor of a component that has an `In` receiver, you can say:

```csharp
this.In.Unsubscribed += closingTime => this.OnUnsubscribe(closingTime)
```

Then in the `OnUnsubscribe` method you can still post messages on your output stream. Note that the `closingTime` in this case will likely be the closing time of the pipeline. 

A good example of this is the [Processor](https://github.com/microsoft/psi/blob/master/Sources/Runtime/Microsoft.Psi/Components/Processor.cs) component, which allows for an `onClose` handler.

KanaHayama, 2024-06-14:
Thank you. The Unsubscribed event and the Processor class are both helpful.

closed 2024-06-14 by KanaHayama