microsoft/psi #319: Error: Invalid Index Entry

KanaHayama, 2024-06-28:
Hello, I am using Psi Store Importer from `0.19.100.1-beta`. For some Psi Stores, I encounter the following error. The Psi Store was captured with the same psi version. In its error message, I do not understand the meaning of `Invalid index entry (extent: 10, position: 224618232, current: 10)`.

Could you help explain why this error might occur and what this error message means? Is there a way to fix the store? How can I check from Psi Studio if the store was closed properly by the writer?

Our Psi Store has many streams, so I am not sure which data type it is handling. Regarding the error, would knowing the stream name and data type be useful in identifying the problem?

```
fail: Microsoft.Extensions.Hosting.Internal.Host[9]
      BackgroundService failed
      System.AggregateException: Pipeline 'default' was terminated because of one or more unexpected errors (Invalid index entry (extent: 10, position: 224618232, current: 10))
       ---> System.ArgumentException: Invalid index entry (extent: 10, position: 224618232, current: 10)
         at Microsoft.Psi.Data.Importer.StreamImporter.Next(Boolean moreDataPromised, Envelope env)
         at Microsoft.Psi.Receiver`1.DeliverNext()
         at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context)
         --- End of inner exception stack trace ---
         at Microsoft.Psi.Pipeline.ThrowIfError()
         at Microsoft.Psi.Pipeline.Run(ReplayDescriptor descriptor, IProgress`1 progress)
         at Microsoft.Psi.Pipeline.Run(TimeInterval replayInterval, Boolean enforceReplayClock, IProgress`1 progress)
         at DataExporter.ExporterService.ExecuteAsync(CancellationToken stoppingToken) in C:\proj\DataExporter\ExporterService.cs:line 69
         at Microsoft.Extensions.Hosting.Internal.Host.TryExecuteBackgroundServiceAsync(BackgroundService backgroundService)
```

sandrist, 2024-06-28:
Are you able to open the store in PsiStudio? Do you get the same exception there?

KanaHayama, 2024-07-01:
I can open the store in PsiStudio. No exception was given in PsiStudio.

KanaHayama, 2024-12-09:
Thank you. That's a good point. I will try to read the store again in a couple of weeks. I will report back.