microsoft/psi #331: [Python Bridging] Issue with Frame Encoding/Byte Conversion

sakibreza, 2024-09-24:
Hi, I’m following the PSI documentation and bridging it with Python to capture and write frames from a webcam. However, I encountered an issue when attempting to convert the frames to bytes for TCP transfer to Python. I’m using the PSI documentation for Python interoperability - https://github.com/microsoft/psi/wiki/Python-Interop-Walkthrough#visualizing-in-python.

The specific error (details provided below) I’m facing occurs during the conversion to bytes (related to encoding). Could anyone help me resolve this issue? Thanks. 

> System.IO.FileNotFoundException
  HResult=0x80070002
  Message=Could not load file or assembly 'PresentationCore, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'. The system cannot find the file specified.
  Source=Microsoft.Psi.Imaging.Windows
  StackTrace:
   at Microsoft.Psi.Imaging.ImageToJpegStreamEncoder.EncodeToStream(Image image, Stream stream)
   at Microsoft.Psi.Imaging.EncodedImage.EncodeFrom(Image image, IImageToStreamEncoder imageEncoder) in Microsoft.Psi.Imaging\EncodedImage.cs:line 81
   at Microsoft.Psi.Imaging.ImageEncoder.Receive(Shared`1 sharedImage, Envelope envelope) in Microsoft.Psi.Imaging\ImageEncoder.cs:line 23
   at Microsoft.Psi.Pipeline.<>c__DisplayClass107_0`1.<CreateReceiver>b__0(Message`1 m)
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass43_1`1.<TrackStateObjectOnContext>b__1()
   at Microsoft.Psi.Executive.PipelineElement.<>c__DisplayClass43_0`1.<TrackStateObjectOnContext>b__0(T m)
   at Microsoft.Psi.Receiver`1.<>c__DisplayClass13_0.<.ctor>b__0(Message`1 m)
   at Microsoft.Psi.Receiver`1.DeliverNext()
   at Microsoft.Psi.Scheduling.Scheduler.ExecuteAndRelease(SynchronizationLock synchronizationObject, Action action, SchedulerContext context)
   at Microsoft.Psi.Scheduling.Scheduler.Run(Object workItem)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()

sandrist, 2024-09-30:
Could you share the details of your project file? Make sure that it is setup as a .NET Framework project, net472. You should see something like the following in your project file:

<TargetFramework>net472</TargetFramework>
<PlatformTarget>x64</PlatformTarget>

and then make sure you are referencing Microsoft.Psi.Imaging.Windows and Microsoft.Psi.Media.Windows.x64

sakibreza, 2024-10-10:
Hi @sandrist, Thanks for the response. It was an issue with .Net version I used. Using net472 solved the issue.

closed 2024-10-10 by sakibreza